adafruit/Adafruit_Blinka_Displayio #12: Implement fill_row in Display class

makermelissa, 2020-05-26:


FoamyGuy, 2020-07-02:
I'm going to make an attempt to implement this over the upcoming weekend. 

While I was looking into a way to take screenshots for Circuit Python devices I realized we already have a library for that: https://github.com/adafruit/Adafruit_CircuitPython_BitmapSaver

I poked around it a little bit and I think if we have `fill_row` implemented and possibly a few other tweaks in the library it may be able to work on Blinka displayio objects.  Then we can have a common screenshot API that works for both, instead of relying on PIL to save an image like I have done now.

makermelissa, 2020-07-02:
Sounds good. I've been doing most of the coding for this by looking are the circuitpython source code and translating it into the Python code except where it has functionality already built into Pillow. In this case, for fill row, you could probably make use of the pillow getpixel() function.

makermelissa, 2020-07-03:
Fixed via #33.

closed 2020-07-03 by makermelissa