adafruit/Adafruit_Blinka_Displayio #9: Add Terminalio support

makermelissa, 2020-05-26:
It would be awesome to be able to display the REPL on the display like with circuitpython. I believe we can just tap into stdout, but it may be more involved than that.

anecdata, 2021-08-16:
Would this include `terminalio.Terminal`?

makermelissa, 2021-08-16:
Yeah, probably. The way that linux and CircuitPython works is a little different though. CircuitPython has the supervisor output to a serial output that it just decodes for the display. For linux, I think we would need to capture the terminal output and redirect to the display. It's similar and I've done it a bit in https://github.com/adafruit/Adafruit_Python_Shell/blob/main/adafruit_shell.py#L66-L101, but I don't think it's the same for what we would need here. It would need to capture Python output only and from the root instance that the library is running in and then you may need to deal with multiple python instances running, etc.