adafruit/Adafruit_Blinka_Displayio #28: Float int TypeError from label simpletest

FoamyGuy, 2020-07-01:
I have this label simpletest modified slightly to work with Blinka and my screen. 
```python
import board
import terminalio
from adafruit_display_text import label
import adafruit_ili9341

# Release any resources currently in use for the displays
displayio.release_displays()

spi = board.SPI()
tft_cs = board.CE0
tft_dc = board.D25

display_bus = displayio.FourWire(
    spi, command=tft_dc, chip_select=tft_cs, reset=board.D24
)
display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240)

text = "Hello world"
text_area = label.Label(terminalio.FONT, text=text)
text_area.x = 10
text_area.y = 10
display.show(text_area)
print("showing")
while True:
    pass
```
When I try to run it I'm getting this error from inside of Blinka displayio:
```
pi@raspberrypi:~/display_layouts $ python3 examples/label_test.py
showing
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/display.py", line 252, in _refresh_loop
    self.refresh()
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/display.py", line 239, in refresh
    self._current_group._fill_area(buffer)  # pylint: disable=protected-access
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/group.py", line 160, in _fill_area
    layer._fill_area(buffer)  # pylint: disable=protected-access
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/tilegrid.py", line 272, in _fill_area
    buffer.alpha_composite(image, (x, y))
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 1544, in alpha_composite
    self.paste(result, box)
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 1496, in paste
    self.im.paste(im, box)
TypeError: integer argument expected, got float
```

makermelissa, 2020-07-01:
Make sure you have the latest code. I believe this was fixed in #27.

FoamyGuy, 2020-07-01:
Ah, thank you! Updated and it is resolved.

closed 2020-07-01 by FoamyGuy