adafruit/Adafruit_Blinka_Displayio #29: Destination must be non-negative

FoamyGuy, 2020-07-01:
I modified the label simple-test again just a bit, removing the x and y positions. 
```python
import board
import terminalio
import displayio
from adafruit_display_text import label
import adafruit_ili9341

# Release any resources currently in use for the displays
displayio.release_displays()

spi = board.SPI()
tft_cs = board.CE0
tft_dc = board.D25

display_bus = displayio.FourWire(
    spi, command=tft_dc, chip_select=tft_cs, reset=board.D24
)
display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240)

text = "Hello world"
text_area = label.Label(terminalio.FONT, text=text)
display.show(text_area)
print("showing")
while True:
    pass
```
Now I am getting this error when the code runs:
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/display.py", line 252, in _refresh_loop
    self.refresh()
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/display.py", line 239, in refresh
    self._current_group._fill_area(buffer)  # pylint: disable=protected-access
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/group.py", line 160, in _fill_area
    layer._fill_area(buffer)  # pylint: disable=protected-access
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/tilegrid.py", line 279, in _fill_area
    buffer.alpha_composite(image, (int(x), int(y)))
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 1523, in alpha_composite
    raise ValueError("Destination must be non-negative")
ValueError: Destination must be non-negative
```
I think this is due to displayio label y origin being in the center. My guess is that PIL is expecting origin (0,0) when it draws and somewhere there is logic to convert from one to the other, but in this case it results in trying to place the label at a negative position which is throwing this error. 

First thought I had to fix it was clamp the positions to 0 and positive values by using `max()` I can give that a try later today and make a PR if I get it working.

jerryneedell, 2020-07-01:
just an unrelated comment -- are you sure you want to be using CE0? I thought that was not recommended for Blinka https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/spi-sensors-devices

FoamyGuy, 2020-07-01:
@jerryneedell hmm that is a good point. I think I started from this example script: https://github.com/adafruit/Adafruit_CircuitPython_ILI9341/blob/master/examples/ili9341_pitft_simpletest.py. I know Melissa helped get me get the pins figured out when I first started testing out Blinka_Displayio a few weeks back, but I don't recall the specifics any more, maybe the example wasn't pushed yet.

Possibly the deal is that we don't have much choice in the CS line in this case because it's on a Pi header expecting CS to be in the "standard" pin location this is the device I'm using: https://www.adafruit.com/product/1601

jerryneedell, 2020-07-01:
I'll admit I find it confusing. I know the LoRa Bonnets use CE1 without any problems ... I just wanted to make sure your were aware of the warnings.

jerryneedell, 2020-07-01:
> for CE0 - you shouldnt use it unless its the _only_ one you use. it is confusing but in this case its the default hardware for the pitft

@ladyada Thanks, that helps clarify things, but why is there any restriction on CE1 as in the note. It works with multiple SPI devices.

ladyada, 2020-07-01:
oh i guess... CE1 should work, as long as CE0 is not being used.

makermelissa, 2020-07-01:
Thanks for finding this @FoamyGuy. I'll give this a try. I'm pretty sure I know how to fix it as I ran into something similar when working on my animated sign. Also, fwiw I've been testing on a PiTFT which uses CE0 for the TFT and CE1 for the STMPE610 touch chip.

jerryneedell, 2020-07-01:
> Also, fwiw I've been testing on a PiTFT which uses CE0 for the TFT and CE1 for the STMPE610 touch chip.

hmm -- I would expect any access of the STMPE610 on CE! to also toggle CE0. Does it work?

makermelissa, 2020-07-01:
> hmm -- I would expect any access of the STMPE610 on CE1 to also toggle CE0. Does it work?

I haven't actually tried using touch, but it initializes fine.

jerryneedell, 2020-07-01:
> > hmm -- I would expect any access of the STMPE610 on CE1 to also toggle CE0. Does it work?
> 
> I haven't actually tried using touch, but it initializes fine.

It may depend on when/what is present on the SPI bus. I'd suggest testing git carefully -- perhaps @ladyada can comment but it looks like a potential problem to me.

closed 2020-07-01 by makermelissa

makermelissa, 2020-07-01:
The only display that I've run into a problem with using CE0 is the RA8875. It just wouldn't work. I had to move CS to a different pin and then it worked.

jerryneedell, 2020-07-01:
> The only display that I've run into a problem with using CE0 is the RA8875. It just wouldn't work. I had to move CS to a different pin and then it worked.

Did it use CE1 for touch? -- I am wondering if the will interfere with CE0

makermelissa, 2020-07-01:
No, it uses a single chip select for bi-directional communications.

jerryneedell, 2020-07-01:
> No, it uses a single chip select for bi-directional communications.

So - the only reason I am concerned is that given the notes in the guide, i would expect CE1 to be active wench any SPI device is toggling its CS -- so if the STMPE is on CE1 it may confuse the TFT on CE0. For that reason, it might be good to try actually using the Touch features just to see if it works properly. If it does we can still wonder why ;-)

PR #30 (2020-07-01): Fixed negative positioning
Fixes #29. I applied the same fix that I used in OpenSign and the label is appearing correctly now.