adafruit/Adafruit_Blinka_Displayio #45: Extra scaling on some Groups

FoamyGuy, 2020-12-31:
I am still working on trying to distill this issue down further to figure out the root cause, but I wanted to get down what I've discovered so far.

I first noticed this issue in the context of display_text labels (and bitmap_labels) so the current example code uses those, but I hope to narrow it down further to just Group, Tilegrid, and Bitmap.

Code:
```python
import displayio
from adafruit_display_text import label
from adafruit_bitmap_font import bitmap_font

# Make the display context
display = # Specific Display setup

splash = displayio.Group(max_size=10)
display.show(splash)
font = bitmap_font.load_font("Helvetica-Bold-16.bdf")

control_text = label.Label(font, text="Hello World!", color=0xFFFF00)
control_text.x = 10
control_text.y = 20

text_group = displayio.Group(max_size=10, scale=2)
group_scaled_text = label.Label(font, text="Hello World!", color=0xFFFF00)
text_group.x = 10
text_group.y = 50
text_group.append(group_scaled_text)

self_scaled_text = label.Label(font, text="Hello World!", color=0xFFFF00, scale=2)
self_scaled_text.x = 10
self_scaled_text.y = 100

splash.append(text_group)
splash.append(control_text)
splash.append(self_scaled_text)
while True:
    pass
```
With this code I think we would expect one small label (the control) and two labels that are 2x the size of the control. One is scaled via it's own scale property, the other is scaled with another Group. I don't think there should be any visible difference between the latter two.

On a PyPortal that is exactly what we get (excuse the blurry photo):
[image]

With Blinka_Displayio the label that is scaled using it's own init parameter seems to be getting scaled an extra time so when `scale=2` it looks visually on the screen like it would if it were actually `scale=4` 

[image]

makermelissa, 2021-01-05:
Hmm, this is definitely inconsistent. I was under the impression that it was supposed to work like in the Blinka version.

@tannewt for scaling multiple groups inside each other, is it supposed to have a compound effect or more of an overriding value effect?

makermelissa, 2021-01-05:
Oh never mind. I see what's going on here. Nice find.

tannewt, 2021-01-06:
> Hmm, this is definitely inconsistent. I was under the impression that it was supposed to work like in the Blinka version.
> 
> @tannewt for scaling multiple groups inside each other, is it supposed to have a compound effect or more of an overriding value effect?

It should compound. The scales should be independent. That means the outer group will have no idea an inner group is scaled.

makermelissa, 2021-01-06:
Thanks, that's what I thought. After looking closer I saw the issue was that the independent groups were affecting each other, which is why I said never mind. :)

kmatch98, 2021-02-07:
I think the above referenced issue and proposed solution may resolve this issue. Once that change gets merged, please check again to see if it is solved.

FoamyGuy, 2021-02-11:
There was a fix for this put into the display_text library that makes the label scaling work the same on Blinka_Displayio and core displayio.

I think there is still something inconsistent between the two, but I never did manage to figure out a more self contained reproducer. 

We could open this back up in the future if something similar comes up. But I'm going to close this for now, display_text was the only example of this issue that I am aware of and the fix there works well for me.

closed 2021-02-11 by FoamyGuy