adafruit/Adafruit_Blinka_Displayio #54: Constructor parameter missing on Display

lesamouraipourpre, 2021-04-21:
There is a parameter missing on the Blinka version of `displayio.Display` specifically `SH1107_addressing`.

```
Traceback (most recent call last):
  File "roomtemp.py", line 41, in configure_display
    display = adafruit_displayio_sh1107.SH1107(display_bus, width=128, height=128)
  File "lib/python3.7/site-packages/adafruit_displayio_sh1107.py", line 73, in __init__
    SH1107_addressing=True,
TypeError: __init__() got an unexpected keyword argument 'SH1107_addressing'
```

The parameter appears to have been renamed from `column_and_page_addressing` in https://github.com/adafruit/circuitpython/commit/06a3d152664a97cc0a2f2faee86d2f5ea4e063e2 which was added in https://github.com/adafruit/circuitpython/commit/f21dc253e094854670b829f6258aac3722eefc26

This renders the [Adafruit_CircuitPython_DisplayIO_SH1107](https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_SH1107) library unusable with Blinka.

My test hardware is the [Pimoroni 1.12" Mono OLED (128x128, white/black) Breakout - I2C](https://shop.pimoroni.com/products/1-12-oled-breakout?variant=29421050757203) on a Raspberry Pi.

PS. I might try to work on a PR, but I don't fully understand the linkage between the C code in adafruit/circuitpython and the Python code in Blinka.

makermelissa, 2021-04-21:
It's failing because monochrome (and grayscale) display support has not been added yet and I'm not seeing an issue for it, so I updated the title.

makermelissa, 2021-04-21:
> PS. I might try to work on a PR, but I don't fully understand the linkage between the C code in adafruit/circuitpython and the Python code in Blinka.

To develop this, I basically looked at the c code and am trying to emulate displayio by providing the APIs and using Pillow for the drawing.

lesamouraipourpre, 2021-04-21:
I'm not sure but I think this may be two different issues then:

1. Support for Monochrome/Grayscale in Blinka
2. Support for SH1107 at the `displayio.Display` level. As far as I can see from the C code, the SH1107 uses a different addressing system to the other ICs supported by `Display` and special checks were put in place for it.

makermelissa, 2021-04-21:
Good point. The SH1107 support was added to CircuitPython since I wrote this, so yep, that's missing.

nkatanaa, 2021-08-13:
Can we try close this one, PLEASE? I want the SH1107 to able run on Raspberry Pi

litui, 2022-03-14:
Looks like this issue is still hanging around. I made a pi-to-featherwing hat for my SH1107 and found quickly that this library sets the `self.SH1107_addressing` variable and that's it. No other instances of `SH1107_addressing` in the code base. I've started poring over the code myself, but between the C and Python libraries the approaches to the display code seem to be different.

PR #114 (2023-09-29): Revamp Displayio and Add grayscale
This change brings Displayio to being much closer in line with the current circuitpython builds, which makes it very accurate, but at this point it is not optimized for running in Python due to some removal of PIL. This adds dirty rectangle tracking as well. I have a bit more testing to do before optimization but it should support monochrome and grayscale OLEDs and allow me to close a number of outstanding issues.

Fixes #105
Fixes #95
Fixes #94
Fixes #54
Fixes #36
Fixes #32
Fixes #7
Fixes #6 
Fixes #4
closed 2023-09-29 by makermelissa