adafruit/Adafruit_Blinka_Displayio #59: Much slower pitft touchscreen performance when using displayio

Jayguitar, 2021-06-18:
Using stmpe610 resistive touchscreen chip with blinka is very fast until the point when the hx8357 screen is initialised using displayio.

I realise they both use SPI but touchscreen responsiveness is dramatically slower compared with using the pitft as a primary display for raspberry pi.

Comparison can be seen by removing hash on line 18 of the code below:

```
import busio
import board
import displayio
import digitalio
from adafruit_stmpe610 import Adafruit_STMPE610_SPI
import adafruit_hx8357

displayio.release_displays()
spi = board.SPI()
tft_cs = board.CE0
tft_dc = board.D25

spi = board.SPI()
cs = digitalio.DigitalInOut(board.CE1)
st = Adafruit_STMPE610_SPI(spi, cs)

display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs)
#display = adafruit_hx8357.HX8357(display_bus, width=480, height=320, backlight_pin=board.D18)

while True:
    if not st.buffer_empty:
        print(st.read_data())
```

Thanks

makermelissa, 2021-06-18:
Yeah, probably because it's not super optimized. #6 would help a lot with this.

makermelissa, 2023-10-02:
It should be better at this point. The main slowdown is it is written in Python rather than c, but it now has optimizations in place.

makermelissa, 2024-01-03:
Closing since there really isn't much more to do on this.

closed 2024-01-03 by makermelissa