adafruit/Adafruit_Blinka_Displayio #71: [WIP] Rewrite of Blinka DisplayIO from CircuitPython core

lesamouraipourpre, 2021-09-19:
# Work in Progress :: Rewrite of Adafruit_Blinka_Displayio from CircuitPython core

Repository/Branch: https://github.com/lesamouraipourpre/Adafruit_Blinka_Displayio/tree/display-core

## Inception

This work started with the intent of translating `display_core.c` from CircuitPython core to Blinka to be used as the superclass of `Display` and `EPaperDisplay`. Without realising it, it quickly escalated into doing a full re-translation of core `displayio` to Blinka.

## Goals

* Phase 1
   * Expose the same API as is exposed from core `displayio` in CircuitPython 7.  _Done, only partially tested_
   * If the API in Blinka is not the same as the API in CircuitPython core, the API in the core takes precedence. Do not care about breaking changes in Blinka DisplayIO. _Done_
   * To translate `vectorio` as part of `displayio`. _Done, only partially tested_
   * To translate `bitmaptools` as part of `displayio`.  _Done, not tested_
* Phase 2 Testing. Testing. And lots more testing.
* Phase 3 Decide how best to speed up the code, by further translating some of it to use `numpy` and maybe `PIL/Pillow`

## Open issues on Blinka DisplayIO
* [Add ePaper Display Support](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/3) : theoretically implemented; testing has failed so far.
* [Finish Shapes Support](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/4) : implemented, and tested. I can find very little example code. Example used from https://github.com/adafruit/circuitpython/issues/3201
* [Add ParallelBus Support](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/5) : the relocation to `paralleldisplay` has been implemented. However, all methods on (Blinka's) ParallelBus currently raise a `NotImplementedError`. This is unlikely to change in the near-term.
* [Implement Dirty Rectangle Tracking](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/6) : implemented. More testing needed.
* [Finish ColorConverter](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/7) : implemented. More testing needed
* [Add Terminalio Support](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/9) : implemented. Currently only stdout is displayed on the screen, NOT stderr. Lots and lots of testing needed.
* [Add Unicode character support to fontio module](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/11) : not implemented. This should be achievable, but I've not attempted it yet.
* [Display warning for zero width/height bitmap](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/32) : will need to be re-tested.
* [Remove thread enabling/disabling for autorefresh](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/36) : may be fixed. The current implementation doesn't attempt to stop the thread. If the `auto_refresh` property is `False`, it sleeps and checks for it being `True` periodically.
* [Add SH1107 display support](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/54) : implemented from the core. Working. More testing needed.
* [Much slower pitft touchscreen performance when using displayio](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/59) : not tested, but it is extremely likely that this will be even slower for now.
* [Update OnDiskBitmap to match core](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/66) : implemented. Tested.
* [remove the max_size parameter from displayio.Group](https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/67) : implemented. I'm assuming it will be weeks or even months before this branch gets merged.

## Testing

### ST7735R (Color SPI) - Working correctly

[0.96" SPI Colour LCD (160x80)](https://shop.pimoroni.com/products/0-96-spi-colour-lcd-160x80-breakout) from Pimoroni. _(Driver ST7735S - using Adafruit_CircuitPython_ST7735R library)_

### SH1107 (B&W I2C) - Working correctly

[1.12" Mono OLED (128x128, white/black) Breakout – I2C](https://shop.pimoroni.com/products/1-12-oled-breakout?variant=29421050757203) from Pimoroni. _(Driver SH1107 - using Adafruit_CircuitPython_DisplayIO_SH1107)_

### SSD1608 / SSD1680 (eInk SPI) - Not working, not sure why
[Inky pHAT Black&White](https://shop.pimoroni.com/products/inky-phat?variant=12549254938707) - (SSD1608)
[Inky pHAT Yellow/Black/White](https://shop.pimoroni.com/products/inky-phat?variant=12549254905939) - (Documented as SSD1608 which is only a B&W eInk controller, I'm assuming it is a SSD1680)

# WHAT NOW?
If you have any hardware running Blinka and an available display to test with, please test the branch below and report:
https://github.com/lesamouraipourpre/Adafruit_Blinka_Displayio/tree/display-core

Please report any successes or failures as an issue on my repository NOT on adafruit/Adafruit_Blinka_Displayio
https://github.com/lesamouraipourpre/Adafruit_Blinka_Displayio/issues
_Note: It is Pure Python, so it WILL BE SLOW. Conversion of parts to NumPy (or alternative) will follow if testing is successful._

I'll try to update the Wiki on my repo as progress is made:
https://github.com/lesamouraipourpre/Adafruit_Blinka_Displayio/wiki

lesamouraipourpre, 2021-09-19:
SH1107 - 128x128
[image]

ST7735S - 160x80
[image]

makermelissa, 2021-09-20:
Thank you for taking this on. When I originally wrote this library, it was based on CircuitPython 5 while the eInk and monochrome display support was still being added to the core, which is why those hadn't been added in and then I was pulled off the project.

I'm not surprised the eInk support isn't quite working as that tends to be the trickiest to work with since a lot of times if it doesn't work, a lot of times they just don't do anything. One way I've debugged eInk displays is to get everything that is being output (possibly using a logic analyzer or oscilloscope). Then get do the same with some kn own working code and try to get that as close as possible. If everything is exactly the same, then take a look at timing and delays.

makermelissa, 2021-12-09:
Hi, what are your plans in regards to the rewrite here? I'm going to work on some of the open issues here and didn't want to duplicate your work too much. Also, it currently uses NumPy an dPillow unless you were referring to your rewrite.

stonehippo, 2022-01-17:
I tried out the current `display-core` with a SSD1306 and the MCP2221A breakout, and it's _close_. The rendering issues I was seeing with the current release (drawing more than once caused some the display to flash, and nothing was rendered correctly), but it looks like the default bitmap font still doesn't display correctly.

I also tried this on Raspberry Pi Zero W. Similar results.

I used [this example code](https://learn.adafruit.com/adafruit-oled-featherwing/python-usage) to test.

SSD1306 with MCP2221
[image]

SSD1306 with Raspberry Pi Zero W
[image]

AlexKlimaj, 2022-03-31:
@lesamouraipourpre I am interested in your rewrite, but I am unable to install it on my Pi. It appears that non of the I2C OLEDs are currently working on Raspberry Pi's with Blinka Circuitpython.

makermelissa, 2023-09-30:
Your work covers a lot of what I've already been doing. I've made use of some of the code from your branch and credited you accordingly. I did take some different directions with regards to the design and much of it is up to date with the latest CircuitPython code. I also have eInks finally working in #115. 

There's still more features that you have in your branch which are still missing in the code here such as vectorio, parallelbus and the supervisor (which I may add to Blinka itself), so I have not closed this issue because of that.