adafruit/Adafruit_Blinka_Displayio #72: Incorrect behavior with SSD1327 via I2C

dkolas-dde, 2021-09-28:
I'm not sure if this would be better reported here or on https://github.com/adafruit/Adafruit_CircuitPython_SSD1327.

When I try to use the SSD1327 via Blinka displayio, the behavior is strange.  I'm using the Adafruit version of the chip: https://www.adafruit.com/product/4741, connected to an FT232H.  The python environment is in Windows.

The display is 128x128, but the way it maps the grid onto the display is not correct.
As best I can tell, the display is acting as though it is only half the width it actually is, despite passing width=128, height=128.

The pixel range y = 0, x=[0:64] is written to the area y=0, x=[0:128].  The pixel range y=0, x=[64:128] is written to y=1, x=[0:64].  This results in twice as many output lines as intended, so the display will write y=[0:64] and then rewrite the whole display with y=[64:128].

The below example, adapted from the various example code, attempts to write a block of 128x64 white to the display, which should cover the top half in white.  What actually happens is it fills the whole display, then wipes it back out.

`import board
import displayio
import adafruit_ssd1327

displayio.release_displays()

# Use for I2C
i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_address=0x3D)

WIDTH = 128
HEIGHT = 128
BORDER = 0
FONTSCALE = 1

display = adafruit_ssd1327.SSD1327(display_bus, width=HEIGHT, height=HEIGHT)

# Make the display context
splash = displayio.Group()
display.auto_refresh = False
display.show(splash)

# Draw a background rectangle, but not the full display size
color_bitmap = displayio.Bitmap(
    #display.width - BORDER * 2, display.height - BORDER * 2, 1
    128, 64, 1
)

color_palette = displayio.Palette(1)
color_palette[0] = 0xFFFFFF  # White
bg_sprite = displayio.TileGrid(
    color_bitmap, pixel_shader=color_palette, x=BORDER, y=BORDER
)
splash.append(bg_sprite)
display.refresh()

displayio.release_displays()`

lesamouraipourpre, 2021-09-28:
The current version of Blinka DIsplayIO has issues with Black & White (and possibly Grayscale) screens.

There is currently a full retranslation of DisplayIO from core CircuitPython to Blinka being worked on. It is being tracked in https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/71

If you are able to, could you checkout the [display-core](https://github.com/lesamouraipourpre/Adafruit_Blinka_Displayio/tree/display-core) branch from my repo and try it out. The main issue with the rewrite so far is speed (or lack of) because it is only in Python. I am currently working on adapting it to use NumPy and/or Pillow for speed. It also occasionally crashes with a `NoneType doesn't have attribute xyz` Error. I may have this fixed but I haven't tested it enough to push the fix yet.

PS. In your code there is an error but it won't affect what is happening to you. Change:
`display = adafruit_ssd1327.SSD1327(display_bus, width=HEIGHT, height=HEIGHT)` to
`display = adafruit_ssd1327.SSD1327(display_bus, width=WIDTH, height=HEIGHT)`

dkolas-dde, 2021-09-28:
@lesamouraipourpre Thanks for the response.  If I replace the library with the display-core branch from the other project, I don't get any code failures/exceptions, but nothing is written to the screen.  Please let me know if there's any other useful debug info I can provide there.

And thanks for noting the width=HEIGHT :)

lesamouraipourpre, 2021-09-28:
I (accidentally) left some debugging code in but commented out.
https://github.com/lesamouraipourpre/Adafruit_Blinka_Displayio/blob/c7c622b7fd8512d5127f5ce59872e38debc99875/displayio
/_i2cdisplay.py#L143-L152

Could you uncomment it and attach a few kB worth of output and I'll have a look at it.

dkolas-dde, 2021-09-28:
Actually - I realized as I was attempting to collect the debug output that it's actually working properly! The program was just exiting before the commands were all being sent to the bus.  

Net result - working properly with the display-core branch.

Thanks for your help!

lesamouraipourpre, 2021-09-28:
That's great. Hopefully, this redevelopment will be finished and merged in the next month or two.

If you get the chance, could you see if this branch correctly handles the different grayscale values and report back. I don't currently have a grayscale display.

dkolas-dde, 2021-09-28:
I tried a range of 128 grayscale values, one per line, which produced a gradient with ~13 discrete grayscale color bands.  I couldn't find in the documentation whether this was the expected amount of differentiation.

[image]

lesamouraipourpre, 2021-09-28:
That looks great, thank you.  The store page for that display says it has 16 levels of grayscale.

makermelissa, 2023-09-20:
Closing in favor of #105.

closed 2023-09-20 by makermelissa