adafruit/Adafruit_Blinka_Displayio #74: SSD1305: "bytearray index out of range"

maksimdrachov, 2021-12-08:
Hi,

I'm having trouble with setting up a SSD1305 OLED display using a raspberry pi.

I'm trying to run the example code as shown in [`adafruit/Adafruit_CircuitPython_DisplayIO_SSD1305`](https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_SSD1305).

However this leads to the following error message:

```
%Run example.py
Traceback (most recent call last):
  File "/home/pi/oled-micropython/example.py", line 25, in <module>
    display = adafruit_displayio_ssd1305.SSD1305(display_bus, width=WIDTH, height=HEIGHT)
  File "/home/pi/.local/lib/python3.9/site-packages/adafruit_displayio_ssd1305.py", line 80, in __init__
    super().__init__(
  File "/usr/local/lib/python3.9/dist-packages/displayio/display.py", line 122, in __init__
    self._initialize(init_sequence)
  File "/usr/local/lib/python3.9/dist-packages/displayio/display.py", line 154, in _initialize
    data_size = init_sequence[i + 1]
IndexError: bytearray index out of range
```

Does anyone have any suggestions? :)

makermelissa, 2022-11-17:
Apparently this is still an issue. Somebody else ran into it.

makermelissa, 2022-11-17:
As for suggestions, I think it's a bug that just needs to be fixed. It wasn't fully tested with all displays, so apparently the SSD1305 just isn't work at the moment, which is one of the reasons we haven't done a 1.0.0 release yet.

closed 2023-06-04 by maksimdrachov