adafruit/Adafruit_Blinka_Displayio #77: TypeError concatenating data_as_command

FoamyGuy, 2021-12-26:
A Discord user mentioned in the #help-with-circuitpython room having an error raised by this line: https://github.com/adafruit/Adafruit_Blinka_Displayio/blob/3354aabaa4481fbd13b8d642bec5ea754d74fc31/displayio/_display.py#L186

https://discord.com/channels/327254708534116352/537365702651150357/924312428844556338

I tested this with Pi 3 B+ and a ili9341 TFT display, I did not get the same error, but looking in the code I'm guessing my display doesn't use data_as_command when sending.

I was able to recreate the same error by aproximating what the displayio code is doing on that line:
```
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> data = b'\x00\x00\x00\xef'
>>> command = 43
>>> bytes([command] + data)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can only concatenate list (not "bytes") to list
>>> 
```
I am looking into a fix for this.

DavidR000, 2022-01-02:
Hi,  I tried the fix by manually editing the file,  while i don't get the error when running the demo code now the display is not showing the hello world demo script but just shows some grid lines,   This is using ssd1327 and the Waveshare 1.5 monochrome oled displa
[image]
y

dhalbert, 2022-01-02:
@DavidR000 notes in discord that the diagonal line is a camera artifact and is not visible to the human eye.

makermelissa, 2022-01-03:
This is probably something simple I missed in my last restructure like not converting the bytes to a list.

PR #78 (2022-01-03): Fix for TypeError send data as command
I think this will resolve #77. However I have not tested the fix on hardware that needs it. I'm using a Pi 3 B+ with ili9341 display. With that environment I think it's not using data_as_command so it's not actually using this line of the driver.

Fix by moving the paren to only covert the command to bytes, then concatenate it with the data which is already type `bytes` 

Tested in python3 REPL on the Pi:
```python
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> [... cut]
closed 2022-01-03 by makermelissa