adafruit/Adafruit_Blinka_Displayio #84: GC9A01 display issue on Raspberry Pi 3 - no text is displayed

mytja, 2022-03-20:
Hello,
So, I bought [one of GC9A01 chip based SPI displays off AliExpress](https://www.makerfabs.com/image/cache/makerfabs/GC9A01%201.28%20Inch%20Round%20LCD%20Module/GC9A01-1.28-Inch-Round-LCD-Module-7-1000x750.jpg) (not a link to AliExpress), and I started testing it. I use CircuitPython with latest `adafruit-blinka` installed on my Raspberry Pi 3B.
So, I installed all necessary libraries:
- [CircuitPython GC9A01](https://github.com/tylercrumpton/CircuitPython_GC9A01)
- [Blinka DisplayIO](https://docs.circuitpython.org/projects/blinka-displayio/en/latest/)

I'm using following code:
```py
import time
import board
import math
import busio
import terminalio
import displayio
from adafruit_display_text import label
import gc9a01

# Release any resources currently in use for the displays
displayio.release_displays()

spi = board.SPI()

# Make the displayio SPI bus and the GC9A01 display
display_bus = displayio.FourWire(spi, command=board.D25, chip_select=board.D18, reset=board.D27)
display = gc9a01.GC9A01(display_bus, width=240, height=240)

# Make the main display context
main = displayio.Group()
display.show(main)

# Draw a text label
text = "Hello\nWorld!"
text_area = label.Label(terminalio.FONT, text=text, color=0xFFFF00,
                        anchor_point=(0.5,0.5), anchored_position=(0,0))
text_group = displayio.Group(scale=2)
text_group.append(text_area) 
main.append(text_group)

# Animate the text 
theta = math.pi
r = 75
while True:
    print(time.monotonic(),"hello")
    text_group.x = 120 + int(r * math.sin(theta))
    text_group.y = 120 + int(r * math.cos(theta))
    theta -= 0.05
    time.sleep(0.01)
```
The problem is, that this code gives me following output (sorry my hands are shaking):

https://user-images.githubusercontent.com/52399966/159176742-e46b4213-cb56-451e-b720-9df9ec32a981.mp4

So, I decided to use Pi Pico to test if display is faulty. So I flashed it with latest CircuitPython firmware, installed everything, modified my code to Pico's pins, and here is the result:

https://user-images.githubusercontent.com/52399966/159176873-305b582f-8b98-4b3d-857b-c8994b6e4a94.mp4

I really don't think it's the [CircuitPython GC9A01](https://github.com/tylercrumpton/CircuitPython_GC9A01)'s fault. I also checked wiring on my Raspberry Pi 3 like a million times. So, my guess is that it's `blinka-displayio`'s fault, as Pico uses CircuitPython's displayio, meanwhile Pi 3 uses `blinka-displayio`, which is a port to Unix & Unix-like operating systems.

I'm also not using integrated (hardware) Chip-Select pin, as it's not recommended.

Let me know what you think is the issue.
Awaiting your response and thank you for everything upfront :pray:

<hr>

[Code source](https://github.com/todbot/CircuitPython_GC9A01_demos/blob/main/examples/gc9a01_helloworld.py)

makermelissa, 2022-06-13:
Currently the Blinka Displayio implementation is not complete and is only a subset of CircuitPython's Displayio. It has not been tested with this display. It's likely using part of the codebase that is missing.

makermelissa, 2023-09-28:
#114 should get it working if you still have the display and are interested. Otherwise, I will likely just close this issue after it's merged.

makermelissa, 2023-10-02:
Closing as this should be fixed.

closed 2023-10-02 by makermelissa