adafruit/Adafruit_Blinka_Displayio #88: Add support for Pillow as (virtual) display

bablokb, 2022-04-16:
I would like to work on supporting Pillow as a sort of virtual display. There are two reasons:

  1. I often work without real hardware, only with my laptop when I'm out of the office. I think the time-consuming layout work could be easily done with Pillow (I could also simulate different display sizes)
  2. Screenshots. Taking photos of real displays usually results in tearing

Pillow has a show() and a save() method, but no way to "refresh the display", so it will be of limited use but maybe there is a workaround for that.

My first idea was to create some special display-bus and display-driver for Pillow, but when I looked at the code of _displayio.py, the complete buffer is already a Pillow-Image, so I wonder if there is a simpler way of doing this.

Any suggestions, hints and tips are welcome, thanks!

FoamyGuy, 2022-04-16:
AFAIK Pillow is primarily an image editing library, it's not really intended for showing the images. There is a function that will open up a window and let you view a still image, but as you noted it's not so straightforward to refresh it with a new version of the image at any sort of reasonable rate. I think you'll have better luck with some sort of graphics rendering engine that is expecting to render something and update it frequently. When I tackled this problem I went with PyGame, but there are likely other graphics frameworks that could handle it as well. 

I created this project that may be of interest to you: https://github.com/FoamyGuy/Blinka_Displayio_PyGameDisplay

There is a class in here that behaves mostly like a normal display as far as Blinka_DisplayIO can tell, but instead of sending the data to a physical display over SPI or I2C it creates a PyGame window that will output whatever the display is showing. 

There is some more recent discussion in this PR about getting it working: https://github.com/adafruit/Adafruit_Blinka_Displayio/pull/63 Blinka_Displayio does not necessarily officially support this use-case (it's is focused more on physical hardware displays) and sometimes changes to Blinka_DisplayIO cause it to stop working. Generally it just needs some exception handling around imports. I don't think too many folks use it so can lag for a bit of time until myself or someone else notices and submits fixes to get it back to running on PCs in this capacity.

bablokb, 2022-04-17:
Thanks for the link, I will have a closer look at that. I know PyGame very well, I have written a python gui framework based on that some years ago.

makermelissa, 2022-06-13:
Closing as I think @FoamyGuy's suggestion is probably your best option.

closed 2022-06-13 by makermelissa