adafruit/Adafruit_Blinka_Displayio #92: Add FontProtocol class to fontio library

rgrizzell, 2022-05-17:
The fontio library for CircuitPython utilizes the `FontProtocol` class for typing. Since this class is not present in this library, Sphinx documentation is failing to build whenever there are references to this class.

```
Running Sphinx v4.5.0
loading intersphinx inventory from https://docs.python.org/3/objects.inv...
loading intersphinx inventory from https://docs.circuitpython.org/en/latest/objects.inv...
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 3 source files that are out of date
updating environment: [new config] 3 added, 0 changed, 0 removed
WARNING: GENERIC_X86 is not fully supported. Some features may not work.                                                                                
reading sources... [100%] index                                                                                                                         

Warning, treated as error:
autodoc: failed to import module 'scrolling_label' from module 'adafruit_display_text'; the following exception was raised:
Traceback (most recent call last):
  File "/home/grizz/.local/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 62, in import_module
    return importlib.import_module(modname)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/grizz/Projects/Adafruit_CircuitPython_Display_Text/adafruit_display_text/scrolling_label.py", line 39, in <module>
    class ScrollingLabel(bitmap_label.Label):
  File "/home/grizz/Projects/Adafruit_CircuitPython_Display_Text/adafruit_display_text/scrolling_label.py", line 56, in ScrollingLabel
    font: FontProtocol,
NameError: name 'FontProtocol' is not defined
```

rgrizzell, 2022-05-17:
I've started out by copying the following snippet and adding it. It resolves the issue with Sphinx, however I feel like there should be something more than just `pass`.

https://github.com/adafruit/circuitpython/blob/1dce806f775bea0b2e4a2cefe0541ebe0dfd433d/shared-bindings/fontio/BuiltinFont.c#L39-L56

FoamyGuy, 2022-05-17:
we can add `circuitpython-stubs` here maybe to get it working: https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/blob/main/docs/requirements.txt

tekktrik, 2022-05-17:
According to PEP 484, `.pyi` files aren't importable, so the above doesn't currently work.  In this case, it probably does make sense to add it somewhere.  Since it relies on `Glyph`, adding it here is a good option, but I also wonder if there's a way to add it to `circuitpython_typing` while keeping `Glyph` in `fontio` here.

tekktrik, 2022-05-18:
@FoamyGuy I also meant to say I tried it and it didn't work 😢