adafruit/Adafruit_Blinka_Displayio #95: SSD1306 displays not working properly

HenryHu, 2022-07-09:
I have a 128x64 display using SSD1306, and it's not working properly.
I noticed that 16384 bytes of data is sent every time, while only 1024 bytes (128 x 64 / 8) should be sent.
I've read some other issues and I understand that this is a known issue. Creating this issue to help others find it.
Can we add some text to the README file to notify future users that monochrome displays are not supported properly yet?

credwards-dynovas, 2022-07-27:
I also have a monochrome SSD1306 128x64 display and it is also not working properly.
In particular, the issue I'm having is that pixels are not displaying as expected. This could be that the extra bytes sent are being misinterpreted - example below:
```
color_bitmap = displayio.Bitmap(WIDTH, HEIGHT, 2)
color_bitmap[1,1]=1
color_palette = displayio.Palette(2)
color_palette[0] = 0x000000
color_palette[1] = 0xFFFFFF
bg_sprite = displayio.TileGrid(color_bitmap, pixel_shader=color_palette, x=0, y=0)
splash.append(bg_sprite)
```
Results in more than 1 pixel turning white. The index of the pixel is also incorrect (see attached photo)
[image]

Potentially related: calling ```sleep()``` on the display results in the following error:
```
Traceback (most recent call last):
  File "/home/dynovas/oled_test/oled_test.py", line 68, in <module>
    try: main()
  File "/home/dynovas/oled_test/oled_test.py", line 11, in main
    circuitpython()
  File "/home/dynovas/oled_test/oled_test.py", line 63, in circuitpython
    oled_display.sleep()
  File "/usr/local/lib/python3.9/dist-packages/adafruit_displayio_ssd1306.py", line 108, in sleep
    self.bus.send(0xAE, b"")  # 0xAE = display off, sleep mode
  File "/usr/local/lib/python3.9/dist-packages/displayio/_i2cdisplay.py", line 86, in send
    self._send(DISPLAY_COMMAND, CHIP_SELECT_UNTOUCHED, bytes([command] + data))
TypeError: can only concatenate list (not "bytes") to list
```
It seems like there may be an issue with how the ```I2CDisplay``` class sends its data

omerk, 2023-07-09:
I think I've just been bitten by this as well, can't get the I2C 128x64 SSD1306 display I have in front of me working on Linux. There appears to be too many writes (screen flickers with seemingly random pixels being lit up), which probably confirms the diagnosis in the first port that there's just too much data being sent to the controller.

If these displays are not indeed supported (deliberate design choice or not) then a note/disclaimer in the README could save serious time/sanity.

Thanks.

PR #114 (2023-09-29): Revamp Displayio and Add grayscale
This change brings Displayio to being much closer in line with the current circuitpython builds, which makes it very accurate, but at this point it is not optimized for running in Python due to some removal of PIL. This adds dirty rectangle tracking as well. I have a bit more testing to do before optimization but it should support monochrome and grayscale OLEDs and allow me to close a number of outstanding issues.

Fixes #105
Fixes #95
Fixes #94
Fixes #54
Fixes #36
Fixes #32
Fixes #7
Fixes #6 
Fixes #4
closed 2023-09-29 by makermelissa