adafruit/Adafruit_Blinka_Displayio #109: ImportError: no module named 'typing' pi pico

Nicchessla, 2023-07-17:
I am trying to run adafruits HX8357 display with the raspberry pico, running micropython using Blinka.
when I import displayio I get the error above

makermelissa, 2023-07-17:
> Is displayio even compatible with Micropython ? It has seemingly never been tested there, has it ? It's not an issue to use typing without guards in a library that is not supported on MP. This should probably be an issue in https://github.com/adafruit/Adafruit_Blinka_Displayio

I don't know why I didn't even think about that sooner. Yeah, this library will not run on MicroPython. It requires the Pillow library and I'm fairly sure you need a single board computer for that.

tekktrik, 2023-07-17:
Is it fair to close this then?  Or should there be any action item?

Nicchessla, 2023-07-18:
> I don't know why I didn't even think about that sooner. Yeah, this library will not run on MicroPython. It requires the Pillow library and I'm fairly sure you need a single board computer for that.

Displayio is part of CircuitPython, yes? So even if I use CircuitPython on my Raspberry Pi Pico it wouldn't work?

tekktrik, 2023-07-18:
Display IO relies on C python libraries, particularly pillow, in order to function. It isn't designed to run on the Pico.

Neradoc, 2023-07-18:
> Displayio is part of CircuitPython, yes? So even if I use CircuitPython on my Raspberry Pi Pico it wouldn't work?

To clarify, you can use displayio if you run Circuitpython instead of Micropython, as it is a builtin library of CP. It's a display and graphics library that needs low level access and optimizations. I don't know what Micropython has to offer as graphics libraries, but I believe it has LVGL bindings that can be compiled in (maybe not by default), which you could use independently of whatever blinka libraries you need.

Nicchessla, 2023-07-18:
I think now I get it. Blinka is really created for single board computers, and not microcontrollers, right?
So, running python with Blinka, displayio would work because of the python libraries mentioned like pillow.
Blinka also works in Micropython on a microcontroller but I guess only very limited then. is this correct?

makermelissa, 2023-07-20:
@Nicchessla Blinka runs fine on MicroPython. It was really this library that was created for SBCs.

makermelissa, 2023-09-13:
Closing as this is pretty much written for linux and not micropython. It would be better to better to either use MicroPython's RGB Display library or go with CircuitPython instead.

closed 2023-09-13 by makermelissa