adafruit/Adafruit_Blinka_Displayio #130: issue with ePaper displays

BlitzCityDIY, 2024-03-11:
hihi- very excited about this refactor! i tried it out with a 7 color ePaper display on the Feather RP2040 ThinkInk with u2if firmware but am getting this error:

```
C:\Users\lizillah\Documents\Learn Guides - Projects\u2if_testing>python 7acep_test.py
Exception in thread Thread-1 (_background):
Traceback (most recent call last):
  File "C:\Users\lizillah\AppData\Local\Programs\Python\Python311\Lib\threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "C:\Users\lizillah\AppData\Local\Programs\Python\Python311\Lib\threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\lizillah\AppData\Local\Programs\Python\Python311\Lib\site-packages\displayio\__init__.py", line 48, in _background
    display._background()  # pylint: disable=protected-access
    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lizillah\AppData\Local\Programs\Python\Python311\Lib\site-packages\displayio\_epaperdisplay.py", line 339, in _background
    if self._ticks_disabled:
       ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ACeP7In' object has no attribute '_ticks_disabled'
Traceback (most recent call last):
  File "C:\Users\lizillah\Documents\Learn Guides - Projects\u2if_testing\7acep_test.py", line 35, in <module>
    pic = displayio.OnDiskBitmap(f)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lizillah\AppData\Local\Programs\Python\Python311\Lib\site-packages\displayio\_ondiskbitmap.py", line 147, in __init__
    palette._set_color(
  File "C:\Users\lizillah\AppData\Local\Programs\Python\Python311\Lib\site-packages\displayio\_palette.py", line 79, in _set_color
    if self._colors[palette_index].rgb888 == color:
       ~~~~~~~~~~~~^^^^^^^^^^^^^^^
IndexError: list index out of range
```

example code:
```
# SPDX-FileCopyrightText: Copyright (c) 2023 Scott Shawcroft for Adafruit Industries
# SPDX-FileCopyrightText: Copyright (c) 2021 Melissa LeBlanc-Williams for Adafruit Industries
#
# SPDX-License-Identifier: Unlicense

'''Simple test for a 5.65" ACeP eInk Display
with the Feather RP2040 ThinkInk'''

import board
import displayio
import busio
import adafruit_acep7in

displayio.release_displays()

# this pinout is for the Feather RP2040 ThinkInk
spi = busio.SPI(board.EPD_SCK, MOSI=board.EPD_MOSI, MISO=None)
epd_cs = board.EPD_CS
epd_dc = board.EPD_DC
epd_reset = board.EPD_RESET
epd_busy = board.EPD_BUSY
display_bus = displayio.FourWire(
    spi, command=epd_dc, chip_select=epd_cs, reset=epd_reset, baudrate=1000000
)

display = adafruit_acep7in.ACeP7In(
    display_bus, width=800, height=480, busy_pin=epd_busy
)

g = displayio.Group()

fn = "C:/Users/lizillah/Documents/Learn Guides - Projects/u2if_testing/display-ruler-720p.bmp"

with open(fn, "rb") as f:
    pic = displayio.OnDiskBitmap(f)
    t = displayio.TileGrid(pic, pixel_shader=pic.pixel_shader)
    g.append(t)

    display.root_group = g

    display.refresh()

while True:
    pass

```

makermelissa, 2024-03-11:
Not terribly surprised. I don't have the 7-color display, so it was never tested despite it being implemented quite differently than other e-inks.

bablokb, 2025-04-29:
I think the problem is not within the ACEP-driver, but within the BlinkaIO epaperdisplay implemention. The variable `_ticks_disabled` is obviously not part of the "contract", i.e. public API. That is why it is prefixed with an underscore. Using it from other code will trigger errors like this.

makermelissa, 2026-09-29:
Fixed by #131 and #172.

closed 2026-09-29 by makermelissa