adafruit/Adafruit_Blinka_Displayio #146: Error in display thread when updating a Label

Neradoc, 2025-01-15:
It's unclear what exactly causes it.
This only happens with certain values of y, it seems 0 and below. 2 doesn't trigger the error for example.
Note that the text is partly off-screen regardless.
Just initing the text_area with y=0 doesn't cause an error.
The sleep seems necessary to trigger the error when y is set.
I also briefly tried to reproduce by moving a simple TileGrid around without success.
```py
import board
import requests
import time
import displayio
import terminalio
import adafruit_displayio_ssd1306

i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_address=0x3D)
oled = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)

splash = displayio.Group()
oled.root_group = splash

from adafruit_display_text import label
text_area = label.Label(terminalio.FONT, text="Test", color=0xFFFFFF, x=0, y=1,)
splash.append(text_area)

time.sleep(1)
text_area.y = 0

while True:
	time.sleep(1)
```
Here is another repro, updating the text.
```py
from adafruit_display_text import label
text_area = label.Label(terminalio.FONT, text="Test", color=0xFFFFFF)
splash.append(text_area)

time.sleep(1)
text_area.text = "Hi"

while True:
	time.sleep(1)
```

```
Exception in thread Thread-1 (_background):
Traceback (most recent call last):
  File ".pyenv/versions/3.11.2/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File ".pyenv/versions/3.11.2/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File ".env/lib/python3.11/site-packages/displayio/__init__.py", line 54, in _background
    display._background()  # pylint: disable=protected-access
    ^^^^^^^^^^^^^^^^^^^^^
  File ".env/lib/python3.11/site-packages/busdisplay/__init__.py", line 332, in _background
    self.refresh()
  File ".env/lib/python3.11/site-packages/busdisplay/__init__.py", line 299, in refresh
    self._refresh_display()
  File ".env/lib/python3.11/site-packages/busdisplay/__init__.py", line 308, in _refresh_display
    self._refresh_area(area)
  File ".env/lib/python3.11/site-packages/busdisplay/__init__.py", line 389, in _refresh_area
    self._core.set_region_to_update(subrectangle)
  File ".env/lib/python3.11/site-packages/displayio/_displaycore.py", line 340, in set_region_to_update
    data += struct.pack(">BB", region_y1, region_y2)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: ubyte format requires 0 <= number <= 255
```

bablokb, 2025-04-29:
I also encountered an error like this. The root cause is probably that there is no locking while the background thread (refresh) uses variables that the foreground thread changes.

jonathanyip, 2025-09-21:
Hey everyone, wanted to share my findings since I was running into the same issue.

I did some debugging and found it might be due to the fact that the library is trying to only do partial updates to the display. With super tiny updates (e.g. like updating a single letter in a label), you end up in a scenario where it tries to update an area smaller than a byte, and you end up with random exceptions because region_y2 is somehow -1 (I'm not an expert on the struct packing, so excuse me for this explanation 😄 )

A solution that worked for me:
1. Turn off auto_refresh

For my specific display, that looked like:
```
self.display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64, auto_refresh=False)
```

2. Manually call display refresh yourself whenever you want to write. But importantly, force a full display refresh:
```
self.display._core.full_refresh = True
self.display.refresh()
```

The screen now reloads nicely without exceptions. The downside is that the entire screen must be rewritten now, but it works well enough for me!

makermelissa, 2026-09-29:
Fixed by #189.

closed 2026-09-29 by makermelissa