adafruit/Adafruit_Blinka_Displayio #153: Area.union() not computed correctly in some cases

FoamyGuy, 2025-01-22:
This code:
```
from displayio._area import Area

area1 = Area(18, 0, 24, 12)
area2 = Area(0, 0, 0, 12)

print(f"area1 = {area1}")
print(f"area2 = {area2}")

area1.union(area2, area1)

print(f"area1 = {area1}")
print(f"area2 = {area2}")
```
Results in this output:
```
area1 = Area TL(18,0) BR(24,12)
area2 = Area TL(0,0) BR(0,12)
area1 = Area TL(0,0) BR(0,12)
area2 = Area TL(0,0) BR(0,12)
```

area1 gets "zero'd out" instead of being modified to contain the full union of the input areas.

This is due to the if statments and early returns here: https://github.com/adafruit/Adafruit_Blinka_Displayio/blob/c35bd2c6474d36ba5f1422dcea74ebe98dcdf8c5/displayio/_area.py#L89-L94

I don't think that I understand what the original intended purpose of those checks and early returns are but I believe in this case it's causing it to copy the empy area into the `union` output area

philrittenhouse, 2025-04-17:
My testing in #158 confirmed this issue.

I don't think it makes sense to copy self into union if self is empty and the same goes for other.   I think the logic is flipped and should probably look like this:

```
        if self.empty():
            other.copy_into(union)
            return
        if other.empty():
            self.copy_into(union)
            return
```

FoamyGuy, 2025-04-17:
resolved by #159

closed 2025-04-17 by FoamyGuy