adafruit/Adafruit_Blinka_Displayio #158: Moving VectorIO shapes leaves old drawing behind Post

philrittenhouse, 2025-04-17:
While trying to reproduce something like the example shown in John Park's CircuitPython Parsec: VectorIO Shapes https://www.youtube.com/watch?v=JjpNlrgTPaQ&t=3s. I am using a Raspberry Pi Zero 2 W and st7789 based display. I have simplified the example to just move a circle across the screen. It draws the circle at each new position but it does not clear the old one.

[image]

Here's my test code:

`import time
import board
import displayio
import terminalio
from vectorio import Circle, Rectangle, Polygon
from fourwire import FourWire
from adafruit_st7789 import ST7789

displayio.release_displays()

spi = board.SPI()
while not spi.try_lock():
    pass
spi.configure(baudrate=24000000) # Configure SPI for 24MHz
spi.unlock()
tft_cs = board.D8
tft_dc = board.D25

display_bus = FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=board.D24)
display = ST7789(display_bus, width=320, height=240, rotation=90)

# Make the display context
main = displayio.Group()
display.root_group = main

circ_palette = displayio.Palette(1)
#circ_palette[0] = (0xFF, 0x55, 0xFF)
circ_palette[0] = 0xFF55FF
ball = Circle(pixel_shader=circ_palette, radius=40, x=120, y=120)
main.append(ball)

cx = 0
while True:
    print(time.monotonic(),"hello world")
    cx += 10
    if cx > 320:
       cx = 0
    ball.x = cx
    time.sleep(0.1)`

philrittenhouse, 2025-04-17:
I found a couple issues in the code that were causing the problem.  The first has already been mentioned in issue #153 

The other issue was in _shape_set_dirty() in _vectorshape.py.  It was only the adding the old position to the _ephemeral_dirty_area causing the circle drawn in the new position to be clipped.

I will create a pull request with the changes I made to fix the issue.

FoamyGuy, 2025-04-17:
resolved by #159

closed 2025-04-17 by FoamyGuy