adafruit/Adafruit_Blinka_Displayio #164: adafruit-blinka-displayio on PiWheels for Python 3.11 is Outdated (v2.3.0), Lacks displayio.FourWire

DroidB, 2025-06-06:
Hello,

When trying to set up an Adafruit ST7789 PiTFT on a Raspberry Pi 5 using the official installer script or manual pip installation, I'm finding it's impossible to get a working displayio environment.

Setup:

Hardware: Raspberry Pi 5 Model B Rev 1.1
OS: Raspberry Pi OS (Bookworm, fresh install)
Python: 3.11.2
Blinka: adafruit-blinka==8.58.1
Problem:
The pip dependency resolver, using the piwheels.org index, installs adafruit-blinka-displayio version 2.3.0. This version provides a displayio module that is missing the essential displayio.FourWire class.

This makes it impossible to use modern CircuitPython drivers like adafruit-circuitpython-st7789 that require displayio.FourWire. We even tried installing directly from the GitHub repository's main branch, and the resulting module still lacked the FourWire attribute.

Is there an updated pre-compiled wheel planned for adafruit-blinka-displayio for Python 3.11 on Raspberry Pi, or a different recommended installation path to get the full displayio API with FourWire on this platform?

Thank you!

FoamyGuy, 2025-06-06:
@DroidB can you please post the code that you are attempting to run? And if it's from a tutorial or example code please link to where you found it. Also it may be helpful if you can copy/paste the exact error message or stack trace that you are getting when you try to run it.

`FourWire` was moved from `displayio` into it's own new module `fourwire`. 

So, current modern circuitpython drivers should not be using `displayio.FourWire` since it was deprecated and removed, they should be using `fourwire.FourWire`. 

I believe that all drivers and example code have already been updated for this, but if you found some reference to `displayio.FourWire` still existing somewhere then we'll want to get that updated. 

There is a PiTFT simpletest here: https://github.com/adafruit/Adafruit_CircuitPython_ST7789/blob/main/examples/st7789_240x135_pitft_simpletest.py#L15 that you can try to run. In this file that it is using `from fourwire import FourWire` instead of displayio.

DroidB, 2025-06-07:
Hello @FoamyGuy , thank you so much for the quick and helpful response!

Your explanation that FourWire was moved to its own fourwire module makes perfect sense. Based on your advice, I set up a completely fresh virtual environment on my Raspberry Pi 5 to try again.

Here is the code I am attempting to run, which includes the from fourwire import FourWire change as you suggested.

Code (final_clock.py):
import time
import board
import displayio
from adafruit_st7789 import ST7789
from adafruit_display_text import label
from fourwire import FourWire
import terminalio
from datetime import datetime, timezone

# --- Release any existing displays to allow this script to take over ---
displayio.release_displays()

# --- Pinout is fixed for the PiTFT HAT, Blinka handles this ---
spi = board.SPI()
tft_cs = board.CE0
tft_dc = board.D25
tft_rst = board.D24

# --- Display Setup for 1.3" 240x240 Display ---
display_bus = FourWire(
    spi, command=tft_dc, chip_select=tft_cs, reset=tft_rst
)

display = ST7789(
    display_bus,
    width=240,
    height=240,
    rotation=90,
    rowstart=0,
    colstart=0
)

# --- Create Display Elements ---
main_group = displayio.Group()
display.show(main_group)
color_bitmap = displayio.Bitmap(display.width, display.height, 1)
color_palette = displayio.Palette(1)
color_palette[0] = 0x000000  # Black
bg_sprite = displayio.TileGrid(color_bitmap, pixel_shader=color_palette, x=0, y=0)
main_group.append(bg_sprite)
y_step = 45
y_start = (display.height - (3 * y_step)) // 2 + 20
local_time_label = label.Label(terminalio.FONT, text="L: HH:MM:SS", scale=3, color=0xFFFFFF)
local_time_label.anchor_point = (0.5, 0.5)
local_time_label.anchored_position = (display.width // 2, y_start)
main_group.append(local_time_label)
utc_time_label = label.Label(terminalio.FONT, text="U: HH:MM:SS", scale=3, color=0x00FF00)
utc_time_label.anchor_point = (0.5, 0.5)
utc_time_label.anchored_position = (display.width // 2, y_start + y_step)
main_group.append(utc_time_label)
date_label = label.Label(terminalio.FONT, text="YYYY-MM-DD", scale=3, color=0x00FFFF)
date_label.anchor_point = (0.5, 0.5)
date_label.anchored_position = (display.width // 2, y_start + (2 * y_step))
main_group.append(date_label)

print("Starting Adafruit DisplayIO clock... Press Ctrl+C to exit.")

# --- Main Loop ---
while True:
    try:
        now_local = datetime.now()
        now_utc = datetime.now(timezone.utc)
        local_time_label.text = f"Local: {now_local.strftime('%H:%M:%S')}"
        utc_time_label.text =   f"UTC:   {now_utc.strftime('%H:%M:%S')}"
        date_label.text =       f"{now_local.strftime('%Y-%m-%d')}"
        time.sleep(1)
    except KeyboardInterrupt:
        print("\nExiting...")
        break

displayio.release_displays()
print("Program finished.")

The main problem now seems to be a packaging issue. Based on your advice, I tried to install the necessary libraries, but pip cannot find the fourwire package.

pip Command and Error:
I ran this command inside the clean virtual environment:

pip install Pillow adafruit-blinka lgpio adafruit-circuitpython-st7789 adafruit-circuitpython-display-text adafruit-circuitpython-displayio-fourwire
This fails with the following error:

ERROR: Could not find a version that satisfies the requirement adafruit-circuitpython-displayio-fourwire (from versions: none)
ERROR: No matching distribution found for adafruit-circuitpython-displayio-fourwire

It seems my RPi5 (Python 3.11, 64-bit OS) cannot find this specific package on PyPI or PiWheels. Any advice on how to correctly install the fourwire module would be greatly appreciated.

Thank you!

FoamyGuy, 2025-06-07:
@DroidB `fourwire` is not an external library that needs to get installed by itself. It's part of Adafruit_Blinka_DisplayIO.

So as long as you have done `pip install adafruit-blinka-displayio` in your virtual environment and it succeeded then you should have the `fourwire` and be able to do `from fourwire import FourWire` in your code.

closed 2025-06-08 by DroidB